ArduPilot/CustomBuild #2: missing features

tridge, 2021-08-10:
Some features for future development:

- [ ] have a button on home page for listing all current builds, with progress, showing board and vehicle type. Allow user to watch the build progress, and download final build when done
- [ ] group build options by category, and expand categories with javascript
- [ ] improved look of site with css in templates

Hwurzburg, 2021-08-11:
-[ ] somehow inform user what to do on fail or put some kind of max memory check for items before build is attempted

davidbuzz, 2021-08-12:
* [ ]  on main screen, give user some sort of estimate on how long it will take, even if this is just "number of items currently in queue x 3" = how many minutes they might wait.  ( as jobs generally take less than 3 minutes ). could be made much smarter, but any estimate is better than none.

hendjoshsr71, 2022-09-27:
Onto the group build options idea which we are doing now.

Add a GUI frontend to allow for common "packages" of features by picture for example.

For easier selection of gimbals have a button to select all of a group.

hyperion11, 2023-05-24:
ideas 
1. Presets to speed up building for upgrading. For example selected_features.json can be used.
2. Add ability to build beta firmware. Now we can build only master and stable.