ArduPilot/CustomBuild #10: can we embedd the enabled features into the .apj or into the hwdef in rom or something.?

davidbuzz, 2021-08-17:


shiv-tyagi, 2025-03-14:
Hey @davidbuzz! Can you please elaborate more? Where are you planning to use that information if we somehow make it happen?

davidbuzz, 2025-03-16:
scenario: 
1 - user makes a custom-build for their flight controller, uses it, finds it vauable.
2 - a lot of time pases, say its a year later.
3 - user re-visits in the future , having completely forgot what options they used to buuild it the first-time.
4 - user cant remember what buld-options were used, wouldnt it be nice if that was stored-away-someplace-safe (eg a .txt file *IN* rom ) so they could download it and re-sue the same settings,?

rmackay9, 2025-03-17:
A simpler solution might be to add "Save config", "Load config" buttons to the left of the "Reset feature defaults" that allows a user to save/load the checked features.  For older builds where some features were not present, the load button would just use the defaults