ArduPilot/CustomBuild #39: Getting continously error message "Sending Unknown message (50)" on 4.3.0

Ltomi78, 2022-09-27:
Issue details
I ve got information from Randy Mackay the reason is there is a coding error in this line in custom fw.
https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L5752
Please describe the problem
I upload 4.3.0 custom build fw https://custom.ardupilot.org/ and now getting this message continously:
"Sending Unknown message (50)"
Setup is: Fc matek f405 wing + gps and compass
Version
4.3.0 build with custom fw builder for matek f405 wing target

Platform
[ ] All
[ ] AntennaTracker
[ ] Copter
[X ] Plane
[ ] Rover
[ ] Submarine

Airframe type
flying wing

Ltomi78, 2022-09-27:
[image]

isoos73, 2022-09-27:
Hi!

I also have this problem. Kakute F7 AIO, plane, 4.2 custom build. 

[image]

Hwurzburg, 2022-09-27:
would need to have the list of build options you used in order to recreate and diagnose....that line has nothing to do  with custom firmware itself...could be that you included adsb when you really dont have that hardware?

isoos73, 2022-09-27:
I checked the "load" parameter in MP on status tab, it shows 4,1-4,4 values. I checked it on my Matek f405 Wing, it has the same values. If I understand correctly, it represents the load of fc's cpu. If it does not hangs the cpu, I do not care, but would be better not to have this annoying message. Thx!

Hwurzburg, 2022-09-27:
I am on vacation this weeek and cant try myself...but will next week...

Ltomi78, 2022-09-27:
> would need to have the list of build options you used in order to recreate and diagnose....that line has nothing to do with custom firmware itself...could be that you included adsb when you really dont have that hardware?

Hi.
I ve also not included adsb but here is my build log about option i ve selected.
After i tried also with much less option but also got this unknown message
[image]

Ltomi78, 2022-09-27:
> I am on vacation this weeek and cant try myself...but will next week...

Thanks for your time..waiting your reply

Hwurzburg, 2022-09-28:
figured out the error and asked dev responsible for it to fix it

Hwurzburg, 2022-09-28:
this has been fixed...try again

closed 2022-09-28 by Hwurzburg

Ltomi78, 2022-09-28:
> this has been fixed...try again

Thank you..for your fast feedback and work..we will check it

Ltomi78, 2022-10-01:
> this has been fixed...try again

All work as it should thank you very much
[image]