ArduPilot/CustomBuild #54: Support beta builds

rmackay9, 2023-06-12:
Currently users can select between Stable and Latest.  It would be nice if during beta testing periods users could also select Beta.

Ryanf55, 2023-11-25:
Or better yet, allow a user to input an arbitrary repo URL and branch number.

The use case is having someone flight-test a PR before it gets merged without requiring someone to manually compile the build.

khancyr, 2023-11-25:
We cannot do open access as that would the open door to abuse and we cannot scale the demand.

If someone have custom code, they should be able to compile it. We can do a need on demand workflow on GitHub action for this and let the person making the request to us its infrastructure

Ryanf55, 2023-11-25:
What about any branch/tag in the repo? Since only maintainers can push branches. 

From earlier discussions, it sounds like we need a way for the custom build server to automatically pick up new betas.

MichelleRos, 2023-11-26:
I quite like the idea of just listing the repos and letting people build any branch from those. Then we could add a few devs' repos, which would allow them to more easily get people to test their code (instead of them having to manually build firmware for each person testing).
Something like a dropdown for the repo and then a text field for the branch?
This would still avoid opening any paths to abuse since people can't build any random code.
Would still be good to make the main Stable & Latest branches easy to find though. Not too sure what would be the best UI for that, though.

Ryanf55, 2023-11-26:
> I quite like the idea of just listing the repos and letting people build any branch from those. Then we could add a few devs' repos, which would allow them to more easily get people to test their [...]

Yea, I had thought of the same thing. Some options include an advanced mode, ordering the options, styling the "recommended" ones differently. That said, the original requests here and in some of the other tickets are specifically at getting branches only on `ardupilot/ardupilot`.

DavideLentini, 2024-02-29:
Hi all , would it be possible to compile a custom build for 4.5.0-beta2 in an alternative method ?Since on the Custom Build Server website it is not possible to compile firmware for beta branches ?

rmackay9, 2024-03-01:
Hi @DavideLentini,

The only other way would be to [setup a build environment](https://ardupilot.org/dev/docs/building-the-code.html) and then build from the [Copter-4.5 branch](https://github.com/ArduPilot/ardupilot/tree/Copter-4.5).

rmackay9, 2024-03-29:
beta builds are apparently now supported!

closed 2024-03-29 by rmackay9