ArduPilot/CustomBuild #58: Copter-4.4.0 with VISO

ZetaMiw, 2023-08-19:
First, I tried to build a light Copter-4.4.0 for KakuteF7MINI.
The result is https://custom.ardupilot.org/builds/copter:KakuteF7Mini:502702df62572519b56971fe9fed5b2883640879:742208b501584972788a1b0daf839c96/
This version works well.

Next, I tried to add VISO to the light Copter-4.4.0 for KakuteF7MINI.
This result is https://custom.ardupilot.org/builds/copter:KakuteF7Mini:502702df62572519b56971fe9fed5b2883640879:7818f7b157a32cc1e88d6f0825f3f18b/
This version workes, but I can not set FRAME type and class. I set Frame type and class by Full Parameter List in Mission Planner, it seems well. MP shows Frame type and class. 
But FC returns following messages,  "Frame: UNSUPPORTED" and " PreArm: Motors: Check frame class and type".

ZetaMiw, 2023-08-19:
Is it the problem of memory?

shiv-tyagi, 2023-08-19:
This is probably not an issue with the firmware builder I think. I see your second build does not include a lot of feature compared to the first one. What frame are you looking for?

ZetaMiw, 2023-08-19:
Thank you for your comment.

I checked and they are not recognized.
Quad: +,x,v,H
Hexa: ;. x.

I want to use Quad H with viso.

Without viso, it works well.

Thank you.

shiv-tyagi, 2025-03-14:
Closing this issue. Thanks.

closed 2025-03-14 by shiv-tyagi