ArduPilot/CustomBuild #66: Can't build Rover 4.4 stable for pixhawk 6x

unavlab, 2023-12-17:
[750/868] Compiling libraries/AP_NavEKF3/AP_NavEKF3_RngBcnFusion.cpp
../build_src/libraries/AP_Baro/AP_Baro.cpp: In member function 'void AP_Baro::init()':
./hwdef.h:559:38: error: 'AP_Baro_ICP201XX' has not been declared
  559 | #define HAL_BARO_PROBE3  ADD_BACKEND(AP_Baro_ICP201XX::probe(*this,GET_I2C_DEVICE(0,0x64)))
      |                                      ^~~~~~~~~~~~~~~~
../build_src/libraries/AP_Baro/AP_Baro.cpp:576:23: note: in definition of macro 'ADD_BACKEND'
  576 |     do { _add_backend(backend);     \
      |                       ^~~~~~~
./hwdef.h:561:61: note: in expansion of macro 'HAL_BARO_PROBE3'
  561 | #define HAL_BARO_PROBE_LIST HAL_BARO_PROBE1;HAL_BARO_PROBE2;HAL_BARO_PROBE3;HAL_BARO_PROBE4
      |                                                             ^~~~~~~~~~~~~~~
../build_src/libraries/AP_Baro/AP_Baro.cpp:641:5: note: in expansion of macro 'HAL_BARO_PROBE_LIST'
  641 |     HAL_BARO_PROBE_LIST;
      |     ^~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.

Hwurzburg, 2023-12-17:
unfortunately, 4.4 does not include that sensor....this may highlight an issue that building older versions with autopilots having  sensor probing causes failures if all the current sensors are not present....

rmackay9, 2023-12-18:
Thanks for the report.  The issue is that we haven't released Rover-4.4 as a stable build, it is still only in beta.  I plan to release it this week though so then this problem should go away.  Now it's possible that a better error message should be displayed though if a branch can't be found.

unavlab, 2023-12-18:
> unfortunately, 4.4 does not include that sensor....this may highlight an issue that building older versions with autopilots having sensor probing causes failures if all the current sensors are not present....

I excluded that sensor in the build settings, hoping that this would bypass the problem..... but it didn't work out ))

Hwurzburg, 2023-12-18:
Turns out the real problem is that there is no rover 4.4 stable yet

unavlab, 2023-12-18:
> Turns out the real problem is that there is no rover 4.4 stable yet

Ничего страшного, останусь пока на предыдущей стабильной версии. Спасибо!

closed 2023-12-18 by unavlab