ArduPilot/CustomBuild #69: Missing ESC Telemetry Option (SpeedyBeeF405Wing)

Magnorio, 2024-02-26:
The ESC Telemetry is not able to run out of any custom compilation becasue the following define is not set:
define HAL_WITH_ESC_TELEM 1
When manually changing the hwdef for the SpeedyBeeF4015Wing and seeting the -> define HAL_WITH_ESC_TELEM 1, then the ESC telemetry is running.
I beleave the other AP boards with 1Mega memory only are suffering from this same issue.

There should be an option to enable this. 

Br,
Pedro Claro

Magnorio, 2024-02-26:
https://discuss.ardupilot.org/t/speedybee-f405-wing-no-esc-telemetry/112650/18?u=pedro_claro

andyp1per, 2024-02-26:
Fixed by https://github.com/ArduPilot/ardupilot/pull/26327

Hwurzburg, 2024-02-27:
we should fix the custom build server issue also.. @peterbarker do you know what the issue is tha esc telem on the Speedybee cant be added back in on the custom build server?