ArduPilot/CustomBuild #70: Unable to save logs on microSD with omnibusF4V2pro

MinaKiyo, 2024-03-06:
The custom build was successful with omnibusF4V2pro, and I was able to write to the flight controller successfully, but I cannot save the log to the microSD.
log_** and log_file_** do not appear in the full parameters of the mission planner.
I want to use Blackbox, but it doesn't work with or without Blackbox.
    I may have made a mistake, but I would be grateful if you could give me some advice.

rmackay9, 2024-03-06:
@MinaKiyo,

We have instructions here for how to [download logs using Mission Planner](https://ardupilot.org/copter/docs/common-downloading-and-analyzing-data-logs-in-mission-planner.html).

Maybe you've already seen this but there are a few odd thing in your comment including "log xxx not appear in the full parameters" because parameters and logs are different things.  Also "Blackbox" is betaflight concept which means the same thing as "onboard log" or "dataflash log".. but I wonder if perhaps you're trying to use a betaflight ground station or something.

MinaKiyo, 2024-03-06:
Thank you for your prompt response.
I haven't been able to tell you the important thing.
A bin file is saved to MicroSD in a "flight coach" on an airplane, and the flight is checked using "Flightplotter".
Plane V4.4.4 OFFICIAL firmware successfully generates a bin file on MicroSD.
I custom built it using the site below as a reference.
https://www.flightcoach.org/wp-content/uploads/Updating-your-FlightCoach-flight-control-board-firmware-V2.pdf
I created "custom build" with "Plane" + "omnibusf4pro" and wrote the firmware, but the bin file is not generated on MicroSD.
The problem occurs regardless of the presence or absence of blackbox, so to keep things simple, let's assume that there is no problem with blackbox.

I apologize for not being able to express my question well.

Hwurzburg, 2024-03-06:
try building with logging enabled under Other...apparently logging got removed by accident on some boards last July...we are working on it

MinaKiyo, 2024-03-06:
Hwurzburg Thank you.
other+logging was not selected.
I'll try testing it tomorrow.
I think this will solve the problem.
I will notify you of the results again.

MinaKiyo, 2024-03-07:
rmackay9　Hwurzburg
thank you.
I solved it by checking other +logging.
Thank you for your support.