ArduPilot/CustomBuild #73: Incorrect warnings`

AbhijithGanesh, 2024-03-19:
## Incorrect warnings
When one installs/clones the repository, the remote may get set to `origin` instead of `upstream` and the user has to manually add the remote and `git fetch` it. Hence there should be a warning instead of an errorr like this.

```python3
/CustomBuild/./app.py:13: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.dir_util import copy_tree
[2024-03-19 22:24:47,441] INFO in app: Got queue lock
[2024-03-19 22:24:47,442] INFO in app: Initial fetch
[2024-03-19 22:24:47,442] INFO in app: Running git rev-parse upstream/master in /base/ardupilot
fatal: ambiguous argument 'upstream/master': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
Traceback (most recent call last):
  File "/CustomBuild/./app.py", line 568, in <module>
    checkout_branch(get_default_branch_name(), s_dir=sourcedir, fetch_and_reset=True)
  File "/CustomBuild/./app.py", line 168, in checkout_branch
    if not on_branch(targetBranch):
  File "/CustomBuild/./app.py", line 151, in on_branch
    git_hash_target = get_git_hash(branch)
  File "/CustomBuild/./app.py", line 148, in get_git_hash
    return subprocess.check_output(['git', 'rev-parse', branch], cwd=sourcedir, encoding='utf-8', shell=False).rstrip()
  File "/usr/lib/python3.10/subprocess.py", line 421, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.10/subprocess.py", line 526, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command '['git', 'rev-parse', 'upstream/master']' returned non-zero exit status 128.
```

Linked issue:
https://github.com/ArduPilot/CustomBuild/issues/72

shiv-tyagi, 2024-03-19:
Hey @AbhijithGanesh! This is how the app is written to work. It _expects_ the remote - upstream set in order to generate builds. It is also mentioned clearly in the setup instructions. In this application, we have to be very specific about the remotes. 
Thanks!

AbhijithGanesh, 2024-03-19:
Apologies for the trouble caused @shiv-tyagi

closed 2024-03-19 by AbhijithGanesh