ArduPilot/CustomBuild #98: Running app.py failed

IvanLeu, 2024-08-13:
I got this error when trying to run ./app.py even though I did all prerequisites for the build
Error message:
 File "/home/leu/ardupilot/custom-build/./app.py", line 547, in <module>
    load_remotes()
  File "/home/leu/ardupilot/custom-build/./app.py", line 90, in load_remotes
    with open(os.path.join(basedir, 'configs', 'remotes.json'), 'r')  as f, open(os.path.join(appdir, 'remotes.schema.json'), 'r') as s:
FileNotFoundError: [Errno 2] No such file or directory: '/home/leu/ardupilot/base/configs/remotes.json'

shiv-tyagi, 2024-08-17:
Hi @IvanLeu! Sorry for a late reply. We have automated a couple of things lately. remotes.json helps in achieving that automation. 

You would need to create a directory named _configs_ inside the base directory.
After this you would need to run `python scripts/fetch_releases.py`. Note that this path is relative to the root of CustomBuild directory. 

This would create the remotes.json file for you.

IvanLeu, 2024-08-18:
> Hi @IvanLeu! Sorry for a late reply. We have automated a couple of things lately. remotes.json helps in achieving that automation.
>
> You would need to create a directory named _configs_ inside the [...]

Thank you for replying. Somehow I missed this script ;) I think this should be added to ReadMe.

closed 2024-08-18 by IvanLeu