ArduPilot/CustomBuild #100: 'Logging' not included in firmware builder 4.5.5 stable

Rolf-G, 2024-08-21:
A friend in a local forum drew my attention to the fact that the custom build for plane 4.5.5 stable 
for the Matek F405-Wing no longer includes ‘Logging’ (to be found behind subitem ‘other’). 

I noticed that ‘Logging’ is also not included even in the MatekF743 custom plane build for 4.5.5 stable as well as in Copter and Rover in contrast to 4.5.4 stable. The current master versions contain ‘Logging’ again. 

For people who want to create a customised 4.5.5 stable, this is certainly unexpected and surprising if logging is not included.  

Rolf
[image]

Hwurzburg, 2024-08-22:
told peterB...he has a fix PR...4.5 firmware builds on firmware server do have logging,even though the extract features say they dont and the build server does not hilite the box as a default

peterbarker, 2024-08-22:
Closed by https://github.com/ArduPilot/ardupilot/pull/27899

shiv-tyagi, 2025-04-15:
Thanks @peterbarker and @Hwurzburg

closed 2025-04-15 by shiv-tyagi