ArduPilot/CustomBuild #102: New vehicle type "Flight Coach"

rolfbartels, 2024-09-06:
I would like to request a new vehicle type call Flight Coach.

We build firmware for the flight coach, and find that there way too many defaults setting enabled for us to have to disable before we can build the firmware.

The Flight Coach Project if found here
https://www.flightcoach.org/

and these are the setting we'd like to see

Default GPS, Compass and Barometer Drivers for selected Flight Controller
Everything else disabled except these.
"EKF2", "EKF3", "CompassLearn", "FixedYawCal", "COMPASS_CAL", "Logging", "PLANE_BLACKBOX"

shiv-tyagi, 2024-09-07:
Hi! The custom build server only serves customised build for standard ardupilot vehicles. Adding a _special_ vehicle of this sort is not technically possible.

shiv-tyagi, 2025-02-03:
I am closing this issue. If you I am misunderstanding something, please comment here and we can open the issue again.

Thanks.

closed 2025-03-24 by shiv-tyagi