ArduPilot/CustomBuild #148: SpeedyBeeF405Wing Copter V4.5.7 failed permanent to compile with or without default settings

Turtletrumpet, 2025-03-19:
Hello,

I try to compile an arducopter V4.5.7 with the frametype OCTAQUAD. After click on "generate build" i get no errormessage in the log, but the compile stopps immediately.

```
Vehicle: Copter
Board: SpeedyBeeF405WING
Remote: ardupilot
git-sha: 2a3dc4b7bf
Version: stable-4.5.7
.
.
.
Enable CRSF Text Param Selection
Enable FrSky Telemetry
Enable FrSkySPort Telemetry
Enable FrSkySPort PassThrough Telemetry
HEXA
OCTA
OCTAQUAD
QUADS(BI,TRI also)
Enable SmartAudio VTX Contol
Enable IRC Tramp VTX Control
Enable VideoTX control

Waiting for build to start...

BUILD_FINISHED
```
example:
[](https://custom.ardupilot.org/builds/copter:SpeedyBeeF405WING:2a3dc4b7bf2507120f7378a7b2fde73185e0c325:ee80f7552f86d6f87ff72cf60d1230dc/)

Please help or give a hint. Thank you.

regards,
Dirk

Turtletrumpet, 2025-03-22:
Hello,
any hints or a solution in sight? Thanks for attention.

regards,
Dirk

Turtletrumpet, 2025-03-26:
Is there hope that a solution can be found?

Hwurzburg, 2025-03-26:
I built with latest and only octoquad successfully....maybe the build option did not exist on 4.5.7

shiv-tyagi, 2025-03-27:
Sorry for a late reply. It doesn't look like a Custom Build Server specific issue. There is probably some problem with the feature-listing/default-features-extraction for the board at stable-4.5.7. I tried the beta-4.6.0 and it built fine. 
Another interesting thing is that the build for the board fails even for the default options selected. 

@Turtletrumpet I would need more time to investigate this.

Turtletrumpet, 2025-03-27:
Hello, same problem with MatekF405-Wing or MatekF405-Wing-bdshot. Could not compile copter V4.5.7 with default options.

Turtletrumpet, 2025-04-08:
Any news or suggestions to solve the problem?

shiv-tyagi, 2025-04-08:
@Turtletrumpet It is a bit weird but the board builds on custom-beta.ardupilot.org :)

Feel free to use it while I figure out whats wrong with the main server.

Turtletrumpet, 2025-04-12:
@shiv-tyagi Hello,
many thanks for the hint. Unfortunately I do not have an environment to compile the firmware myself. I will wait for your solution. Thanks again for your efforts.

shiv-tyagi, 2025-04-12:
@Turtletrumpet 
No worries. Feel free to use custom-beta as I mentioned above. It works there.

Turtletrumpet, 2025-04-12:
I give your instructions a try ;-). I installed debian 12 in a virtual machine and follow your instructions....

Turtletrumpet, 2025-04-12:
@shiv-tyagi I use ubuntu 24.04. Should I install with graphical interface or is there no advantage?

Turtletrumpet, 2025-04-12:
Ok. It works,

[image]

shiv-tyagi, 2025-07-08:
This is fixed on custom.ardupilot.org

closed 2025-07-08 by shiv-tyagi