ArduPilot/CustomBuild #157: Header include issue breaking build: AP_Baro_MS5611.cpp

Demine-James, 2025-04-18:
Hi Team, 
Issue compiling on the custom build server, prob header include block typo :)
Trying to build for H473 with boot loader.

'[ 901/1232] Compiling libraries/AP_NavEKF3/AP_NavEKF3_VehicleStatus.cpp
../build_src/libraries/AP_Baro/AP_Baro.cpp: In member function 'void AP_Baro::init()':
./hwdef.h:655:38: error: 'AP_Baro_MS5611' has not been declared
  655 | #define HAL_BARO_PROBE1  ADD_BACKEND(AP_Baro_MS5611::probe(*this,GET_I2C_DEVICE(0,0x77)))
      |                                      ^~~~~~~~~~~~~~
../build_src/libraries/AP_Baro/AP_Baro.cpp:521:23: note: in definition of macro 'ADD_BACKEND'
  521 |     do { _add_backend(backend);     \
      |                       ^~~~~~~
./hwdef.h:658:29: note: in expansion of macro 'HAL_BARO_PROBE1'
  658 | #define HAL_BARO_PROBE_LIST HAL_BARO_PROBE1;HAL_BARO_PROBE2;HAL_BARO_PROBE3
      |                             ^~~~~~~~~~~~~~~
../build_src/libraries/AP_Baro/AP_Baro.cpp:589:5: note: in expansion of macro 'HAL_BARO_PROBE_LIST'
  589 |     HAL_BARO_PROBE_LIST;
      |     ^~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.

Waf: Leaving directory `/mnt/volume_nyc3_01/custom/base/tmp/copter:MatekH743-bdshot:1a0e997ac25eeedcf34c8e8d4364cad7c1555b15:7e93ac3e8eeb737a9627283913f9e83f/MatekH743-bdshot'
Build failed
 -> task in 'objs/AP_Baro/ArduCopter' failed (exit status 1): 
	{task 139991294767200: cxx AP_Baro.cpp -> AP_Baro.cpp.3.o}
 (run with -v to display more information)
done build

BUILD_FINISHED`

peterbarker, 2025-04-20:
@Demine-James until my PR is merged you can get manually tick the MS5611 support in the interface to avoid this error.

closed 2025-04-20 by peterbarker

peterbarker, 2025-04-20:
@Demine-James should be fixed

Demine-James, 2025-04-20:
Thank you Peter <3 !