ArduPilot/CustomBuild #171: Add a build: Select Version could default to "Stable"

rmackay9, 2025-06-07:
I think when the custom build server was first released, it only worked with "latest" but now that it's matured we should make "stable" the default instead of "latest".  It's just about setting the defaults to what we'd like users to use.

I think the order should be (if possible)

- stable
- beta
- latest

Because we have multiple stable builds, I guess beta and latest could be items 2 and 3 but I don't feel too strongly about this.

shiv-tyagi, 2025-06-07:
I totally agree with you @rmackay9. I had this thought in my mind I was just waiting for someone else to raise this to make sure I am thinking correctly. 

I just created #172.

Hwurzburg, 2025-06-07:
I personally dislike this...it should be IMHO:
current stable
beta
latest
all other older stable releases
tags

Hwurzburg, 2025-06-07:
Most should not be using old stables unless a very specific need, so I still disagree

PR #172 (2025-06-14): web: fix versions sorting
Closes #171 

Show the stable versions first followed by beta and then the latest.

Do proper versions comparison instead of simple lexicographical string matching.

The new ordering would look like,
[image]
closed 2025-06-14 by shiv-tyagi